ppannuto/python-saleae #5: Python 2.7 support

LeonardMH, 2015-09-28:
There is no mention of this in the README so I'm guessing it's unsupported but I'm unable to get this package working in Python 2.7.10 (which I unfortunately have to use).

I installed through pip and initially had an import error where it was failing to `import enum` so then I did a `pip install enum` and tried again but the script is now failing with `AttributError: 'module' object has no attribute 'unique'` in reference to the enum module.

ppannuto, 2015-09-28:
Yeah, I didn't make any effort towards backwards compatibility when I wrote
this. At the same time there isn't a ton of code. I'm happy to take a pull
request that adds py2k support, but it's not something I have any interest
in doing. Sorry.

LeonardMH, 2015-09-29:
Understandable, I wouldn't either if I didn't have to. With that said, it doesn't look like too difficult of a change to make. Would you be opposed to a pull request that converts the enums to dictionaries? A few of the other changes I came across are:

1. A try-except block for the bytes() function, in Python 2.7 this function doesn't have the format argument and I believe it defaults to UTF-8 anyways.
2. In the demo function, do an initial check of the Python version, if it is less than 3 set input = raw_input. In Python 2.7 the input function did something very different but the raw_input function behaves essentially the same.

ppannuto, 2015-09-29:
> Sounds good
>    1. In the demo function, do an initial check of the Python version, if
> I've done that before w/out issue. Sounds good.

> Understandable, I wouldn't either if I didn't have to. With that said, it
> doesn't look like too difficult of a change to make. Would you be opposed
> to a pull request that converts the enums to dictionaries?
Don't feel strongly about this, but anything wrong with just adding a
dependency on the enum34 package? Less work for you.
http://stackoverflow.com/questions/36932/how-can-i-represent-an-enum-in-python

LeonardMH, 2015-09-29:
Good find on enum34, I had installed enum which would explain why it isn't working at all. I'll check that out.

closed 2015-12-01 by ppannuto

LeonardMH, 2015-12-02:
`ConnectionRefusedError` is still undefined in Python 2.7.

Not exactly a showstopper, but it can be misleading to newbies.