ppannuto/python-saleae #8: Issues with Logic 8

echiou, 2015-10-30:
When using a Logic 8 (both in demo mode and when one is plugged into the computer), `self.set_sample_rate_by_minimum()` asserts that `self.sample_rates[0][0] >= self.sample_rates[-1][0]`, and it seems like the Logic 8 does the opposite (it returns it sorted lowest to highest). Is this intentional, or a bug somewhere (and thus we should check for which device we are using).

ppannuto, 2015-11-09:
Hmm.. I'm not sure about this. @Marcus10110 is there an expected behavior here?

Marcus10110, 2015-11-09:
There is nothing in the software to guarantee that the sample rates are sorted, but in general, I would expect them to be sorted from analog slowest to analog fastest. (this implies digital fastest to digital slowest, but due to complex FPGA requirements, you will find frequent exceptions to this)
It's possible that analog sample rates are always sorted from fastest to slowest.

Basically, for any given performance setting (which directly relates to USB throughput) the available sample rates try to offer a trade off between digital and analog sample rates. (i.e. trade analog bandwidth for digital bandwidth, and vice versa)

I would update set_sample_rate_by_minimum to scan all performance options, and every sample rate option. I would ignore sorting and check every option, while tracking the lowest analog & digital pair that satisfies the requirement.

if you would like to select exactly the slowest sample rate for a given requirement, the USB bit rate can be calculated like this:

ADC width = 12 bits for Logic 8, Pro 8 and Pro 16.
ADC width = 8 bits for logic 4.

Bandwidth (bits per second) = (digital_sample_rate * digital_channel_count) + (analog_sample_rate * analog_channel_count * adc_width). 

That's the exact USB bandwidth requirement for any sample rate combination. There is no additional overhead. The lower the bandwidth, the higher the reliability for long duration captures. (this really only affects the original Logic and Logic 4, and only on poor-performance host controllers)

commit d785fab8 (2015-11-19): Exhuastively search for minimum sample rate

Available ample rates are not necessarily sorted, so we need to check
all of them to find the best. We sort this by USB bandwidth. Currently
this will hit Saleae several times to ask for the connected devices and
active channels, but I don't think that's a big deal in practice and
makes `get_bandwidth` more accessible for other uses.

Closes #8.
closed 2015-11-19 by ppannuto