ppannuto/python-saleae #9: Logic16 CommandNAKedError on save_to_file while in loop

bacaj1, 2015-11-11:
I am attempting to write a test utility which will capture and save a new waveform every 20 seconds. After the first 2-3 successful iterations in the loop, the SaveDataToFile routine is raising CommandNAKedError which to my understanding of the docs means that the capture isn't complete when called or there is an issue with the filepath. As the first few files saved successfully, I doubt that the filepath is the issue, though watching the GUI the capture is definitely finished when I call save_to_file

I have attempted to wrap your library in a class of my own to allow for easy reuse. 

I was having trouble with the capture_start_and_wait_until_finished function, so I took the code in it and built my own which catches the NAK exception and ignores it. Is this the correct method for dealing with the NAK when calling this function before capture is complete? 

I'm using Python 3.5 / Logic 1.2.5

Here is the loop I am currently attempting to run and my class

```
l.StartCapture()
l.WaitUntilCaptureComplete()
l.SaveDataToFile(filepath + "\Error_" + str(count) + ".logicdata")
time.sleep(20)

```

```
class LogicAnalyzer:

    s = 0

    # Initializes the Saleae API
    def __init__(self, host='localhost', port=10429):
        self.s = saleae.Saleae(host,port)

        print("Connected to Logic")

        return

    # Starts the capture sequence
    def StartCapture(self):
        print("Starting capture")
        self.s.capture_start()
        return

    def WaitUntilCaptureComplete(self):
        done = False
        while done == False:
            try:	
                while not self.s.is_processing_complete():
                    print("\t..waiting for capture to complete")
                    time.sleep(1)

                print("Capture complete")

                done = True

            except self.s.CommandNAKedError:
                # This exception is raised if processing of capture is not complete on the
                # side of the analyzer. This is expected for the first few calls, therefore we 
                # simply ignore it until is_processing_complete returns true
                pass

    # Will save last capture in given output location/file
    # Filename must be absolute path
    # Folder must exist for file to be written
    def SaveDataToFile(self, filename):

        self.s.save_to_file(filename)
        print("File Saved")
        return

```

Not sure if it will help but I also made a log as requested in analog-io's thread. 
[logfile.txt](https://github.com/ppannuto/python-saleae/files/32217/logfile.txt)

bacaj1, 2015-11-12:
I modified my loop to use capture_to_file instead, and that appears to be working (through 25 iterations). 

I'd still like to know what was wrong with my previous implementation if you have time to look into it

ppannuto, 2015-11-12:
> I was having trouble with the capture_start_and_wait_until_finished
function, so I took the code in it and built my own which catches the NAK
exception and ignores it. Is this the correct method for dealing with the
NAK when calling this function before capture is complete?

Ignoring is fine so long as you keep waiting. NAK in this context is "not
done yet", so you need to keep looping until you get an ACK.

closed 2015-11-19 by ppannuto