ppannuto/python-saleae #16: Error running demo()

manuelgodoy, 2016-03-11:
I am trying to run the demo and I get the following error

```python
>>> saleae.demo()
Running Saleae connection demo.

Saleae connected.
Press Enter to continue...

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/saleae/saleae.py", line 493, in demo
    s.set_performance(PerformanceOption.Full)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/saleae/saleae.py", line 267, in set_performance
    self._cmd('SET_PERFORMANCE, {}'.format(performance.value))
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/saleae/saleae.py", line 112, in _cmd
    self._send(s)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/saleae/saleae.py", line 99, in _send
    self._s.send(bytes(s + '\0', 'UTF-8'))
TypeError: str() takes at most 1 argument (2 given)
```

manuelgodoy, 2016-03-11:
Actually I just noticed your code is for Python 3. Do you plan to make it compatible with Python 2.7?

ppannuto, 2016-03-13:
Someone else had an issue a little while back and I added some things that were able to fix Python2 support, see #5. I'm happy to merge in fixes for Py2k, but don't have the time to develop it myself.

Unfortunately, this isn't the easiest thing to add automated testing to since it requires an external program to be running.

Any particular reason you're still using Py2k? It is deprecated...

closed 2016-03-13 by ppannuto

ppannuto, 2016-03-14:
I did just test this locally with Python 2.7.11 and the demo seems to work fine.

Are you sure your installation is up to date, https://github.com/ppannuto/python-saleae/blob/master/saleae/saleae.py#L9 should fix this issue

```python
Python 2.7.11 (default, Jan 11 2016, 00:48:43)
[GCC 4.2.1 Compatible Apple LLVM 7.0.2 (clang-700.1.81)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> type(bytes)
<type 'type'>
>>> bytes == str
True
>>> from builtins import bytes
>>> bytes == str
False
>>> type(bytes)
<class 'future.types.newbytes.BaseNewBytes'>
```

manuelgodoy, 2016-03-18:
@ppannuto thanks for your replies. I am using Python 2.7 because many libraries use it and a many of my current projects were built in it. I decided to work with a virtual environment to use Python 3 and saleae.py. In any case my demo() never worked with Python 2.7 but I am not going to troubleshoot that any time soon.