ppannuto/python-saleae #22: set_active_channels does not send a valid command to Logic

mauzybroadway, 2016-08-11:
Passing a list argument to saleae.set_active_channels() fails with the following error:

`File "C:\PA_Tools\Python27\lib\site-packages\saleae\saleae.py", line 109, in _finish`
`ret = self._cmd(', '.join(self._to_send))`
`TypeError: sequence item 2: expected string or Unicode, list found`

Simple fix is to just `','.join(['{0:d}'.format(ch) for ch in digital])` on line 483 and `','.join(['{0:d}'.format(ch) for ch in analog])` on line 486.

ppannuto, 2016-08-16:
How are you calling this API, it seems to work correctly for me?

```
$ python2
Python 2.7.12 (default, Jun 29 2016, 14:05:02)
[GCC 4.2.1 Compatible Apple LLVM 7.3.0 (clang-703.0.31)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import saleae
>>> s = saleae.Saleae()
INFO:saleae.saleae:Connected.
>>> s.set_active_channels([0,1,2,3], [0])
DEBUG:saleae.saleae:Send >SET_ACTIVE_CHANNELS, digital_channels, 0,1,2,3, analog_channels, 0<
```

ppannuto, 2016-09-13:
Closing because inactive. Feel free to re-open with more info if this is still an issue.

closed 2016-09-13 by ppannuto