ppannuto/python-saleae #33: Error running on windows

orbitinstasis, 2017-04-11:
Connected devices:
        <saleae.ConnectedDevice #1 LOGIC_4_DEVICE Logic 4 (4d7e795b740921b3) **ACTIVE**>
Only one Saleae device. Skipping device selection
Press Enter to continue...

Setting active channels (digital=[0, 1, 2, 3, 4], analog=[0, 1])
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\benka\Desktop\python-saleae-master\saleae\saleae.py", line 959, in demo
    s.set_active_channels(digital, analog)
  File "C:\Users\benka\Desktop\python-saleae-master\saleae\saleae.py", line 556, in set_active_channels
    self._finish()
  File "C:\Users\benka\Desktop\python-saleae-master\saleae\saleae.py", line 167, in _finish
    ret = self._cmd(', '.join(self._to_send))
  File "C:\Users\benka\Desktop\python-saleae-master\saleae\saleae.py", line 200, in _cmd
    ret = self._recv(expect_nak=expect_nak)
  File "C:\Users\benka\Desktop\python-saleae-master\saleae\saleae.py", line 191, in _recv
    raise self.CommandNAKedError
saleae.saleae.CommandNAKedError

ppannuto, 2017-05-13:
Hey, sorry I didn't see this when it came in.

I don't have a Windows machine, so this is a bit tricky for me to debug.

Actually, I don't think this has anything to do with Windows:
[image]

It looks like the demo tries to set more active channels than the Logic 4 has available, which is why that fails. I can update the demo to fix that

orbitinstasis, 2017-05-13:
ahh. I guess that explains it.

Ben

commit b5c4eac9 (2017-05-13): don't try to or allow setting channels on Logic 4

Looks like Logic 4 doesn't support setting channels over the scripting server:
https://github.com/saleae/SaleaeSocketApi/blob/master/SaleaeSocketApi/SocketApi.cs#L899

Closes #33.
closed 2017-05-13 by ppannuto