ppannuto/python-saleae #42: parameter in command set_trigger

liuchaoatsz, 2018-07-17:
Hi,Pannuto
       I'm trying to use saleae to do something .
      I try the function **set_trigger_one_channel**  and I find out that I am not able to use it to set the channel int Trigger.High/Trigger.Low mode.but Trigger.Posedge/Trigger.Negedge is ok.
       I check the API docs in **saleae/SaleaeSocketApi/Doc/Logic Socket API Users Guide.md** and I found that Python version don't implement the  **SET_TRIGGER** command competely.
       Did I misunderstand this point ?
       Looking forward to your reply  .

ppannuto, 2018-07-17:
I believe that [saleae/SaleaeSocketApi/Doc/Logic Socket API Users Guide.md](https://github.com/saleae/SaleaeSocketApi/blob/master/Doc/Logic%20Socket%20API%20Users%20Guide.md) describes only the C# API. Saleae does not maintain the Python API, just me in my spare time :).

The Python API does implement the `SET_TRIGGER` command. The Saleae protocol is a little awkward, so we expose [a convenience method to set one channel](https://python-saleae.readthedocs.io/en/latest/#saleae.Saleae.set_trigger_one_channel), or [`set_triggers_for_all_channels`](https://python-saleae.readthedocs.io/en/latest/#saleae.Saleae.set_triggers_for_all_channels) to set them all, which matches the C# `SetTrigger` method.

The Python API currently supports `NoTrigger`, `High`, `Low`, `Posedge`, and `Negedge` trigger types.

liuchaoatsz, 2018-07-18:
Hello ppannuto:
         Thanks for your reply.
         This project is very helpful. Thanks for your efforts.
         I mean we couldn't to specify a pulse gate like below:

> 4 channels, with a positive pulse on channel 0, no other channels used by the trigger. The pulse must be between 1ms and 2ms in duration.
> set_trigger, pospulse, 0.001, 0.002,,,
        Quote from   [saleae/SaleaeSocketApi/Doc/Logic Socket API Users Guide.md](https://github.com/saleae/SaleaeSocketApi/blob/master/Doc/Logic%20Socket%20API%20Users%20Guide.md)

liuchaoatsz, 2018-07-18:
Hello  ppannuto,
Thanks for your reply.
I‘d like to try to implement it .

Renegade85, 2018-08-13:
Hi ppannuto,

I could try to implement the new features as well. The only problem is, that the pospulse/negpulse requires two new options to be included - the min and max time value. So if you don't mind, I would extend the current set_triggers_for_all_channels() by a new parameter (tuple) pulse_width_min = 0 and pulse_width_max which would be set to a default value. This could guarantee backwards compatibility with older scripts. If you prefer a different input argument, just let me know your opinion. :-)

Have a nice day,
ren

Renegade85, 2018-08-16:
Sounds reasonable to me. Although same things are implemented differently - current triggers are enums, while trigger pulses will be represented by a class, which feels not very clean.

I'm from C world and in Python just a beginner, that's why I proposed my solution that way. Also one part is I don't like the no-type in Python, as usually it's hard to guess, what an input can be. Nevertheless, this is not your case, as your code is well documented in this respect and one knows, which inputs are required.

Maybe it would make sense to implement the triggers as one "type".

I'll implement it based on your suggestion and we will see, it can be fixed in the future if required.

ppannuto, 2020-02-20:
n.b.: partial implementation in #43

ppannuto, 2021-08-16:
I believe this is addressed via #79.

closed 2021-08-16 by ppannuto