ppannuto/python-saleae #49: Add parameter to launch_logic allowing the caller to specify path

LeonardMH, 2019-03-18:
It would be useful to be able to specify the path to the Logic application when calling `launch_logic` so that users with Logic installed in a non-standard location can still use this method.

Suggested change:

```python
@staticmethod
def launch_logic(timeout=5, quiet=False, logic_path=None):
    '''Attempts to open Saleae Logic software'''
    if platform.system() == 'Darwin':
        logic_path = logic_path or '/Applications/Logic.app'
        if os.system('open {}'.format(logic_path)) != 0:
            raise OSError("Failed to open Logic software")
    elif platform.system() == 'Linux':
        if quiet:
            mode = ' > /dev/null 2>&1 &'
        else:
            mode = ' &'

        if logic_path is not None:
            os.system(logic_path + mode)
        elif PY2K:
            log.warn("PY2K support limited. If `Logic` is not on your PATH it will not open.")
            os.system("Logic" + mode)
        else:
            path = shutil.which('Logic')
            if path is None:
                raise OSError("Cannot find Logic software. Is 'Logic' in your PATH?")
            os.system(path + mode)
    elif platform.system() == 'Windows':
        if logic_path is not None:
            p = logic_path
        else:
            p = os.path.join("C:", os.sep, "Program Files", "Saleae Inc", "Logic.exe")
            if not os.path.exists(p):
                p = os.path.join("C:", os.sep, "Program Files", "Saleae LLC", "Logic.exe")
        os.startfile(p)
    else:
        raise NotImplementedError("Unknown platform " + platform.system())

    # Try to intelligently wait for Logic to be ready, but can't wait
    # forever because user may not have enabled the scripting server
    while timeout > 0:
        with contextlib.closing(socket.socket(socket.AF_INET, socket.SOCK_STREAM)) as sock:
            if sock.connect_ex(('localhost', 10429)) == 0:
                break
        log.debug('launch_logic: port not yet open, sleeping 1s')
        time.sleep(1)
        timeout -= 1
```

ppannuto, 2019-03-19:
Looks reasonable to me. Mind converting this to a PR? Also, if you could add documentation on the new parameter (oy, and apparently the existing parameters!) that would make this a slam dunk.

For the existing ones that don't have documentation.... something like this:
```python
'''Attempts to open Saleae Logic software.

:param timeout: Time in seconds to wait for the Logic software to launch
:param quiet: Silence terminal prints from Logic software
'''

LeonardMH, 2019-03-20:
Sure, I'll open a PR later today.

LeonardMH, 2019-03-20:
@ppannuto, PR #50 opened.

ppannuto, 2019-03-21:
Fixed by #50.

closed 2019-03-21 by ppannuto