ppannuto/python-saleae #51: export_analyzer returns before csv file is completed

kubaraczkowski, 2019-05-08:
Hi,

I'm seeing an erratic behavior when running loops around the export_analyzer functionality.
As far as I see the function returns before the CSV file is complete - this means that subsequent reads of the csv return less data than intended.

Is that something on the python-wrapper level though, or a trouble with the C++ API?
Or - obviously - am I doing sth wrong?

ppannuto, 2020-02-20:
Sorry! I somehow missed this. In the Saleae documentation (which is largely copied over into the docstrings) they note:

> Export analyzer index N and save to absolute path save_path. **The analyzer must be finished processing** _(emphasis mine)_

Are you doing anything to check that it's complete (i.e. calling `is_analyzer_complete`) first?

kubaraczkowski, 2020-02-20:
Gosh, it has been a while now, we actually had to move to Digilent's system
to avoid hitting this issue.

The problem was indeed that the analyzer is finished
(is_analyzer_complete() is true) but the exported CSV is not closed or
flushed. This results in a reading error when trying to load that CSV...

ppannuto, 2020-02-20:
Sorry to hear that. I'm guessing from the sounds of it this was a larger file. I think the `export_analyzer` method just triggers the write to start, but doesn't have any way to report whether the write was complete. I'm guessing a small `sleep` afterwords would have worked, but been a bit dissatisfying.

I'll ping @Marcus10110 in case this is something Saleae is interested in knowing about / has a solution, but I'll probably also close this issue here as I don't think there's anything the Python wrapper can do. Feel free to re-open (anyone) if there's something more that this library can do here.

closed 2020-02-20 by ppannuto