ppannuto/python-saleae #52: Random CommandNAKedError occurrence during "capture_to_file(log_file)" call

drchopper, 2019-11-12:
I created a small python application which shall sample my hardware during boot. Therefore, I start the board every second overnight.
Unfortunately, my application raise a CommandNAKedError already after my approx. 10 reboots. Until the error occurs, the script works as expected.

Any idea how to get them stable?

```
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "av_kit_mdio_analyzer.py", line 35, in logic_task
    device.capture_to_file(log_file)
  File "/localdisk/twesterm/projects/AVKIT/venv/local/lib/python2.7/site-packages/saleae/saleae.py", line 711, in capture_to_file
    self._cmd('CAPTURE_TO_FILE, ' + file_path_on_target_machine)
  File "/localdisk/twesterm/projects/AVKIT/venv/local/lib/python2.7/site-packages/saleae/saleae.py", line 257, in _cmd
    ret = self._recv(expect_nak=expect_nak)
  File "/localdisk/twesterm/projects/AVKIT/venv/local/lib/python2.7/site-packages/saleae/saleae.py", line 248, in _recv
    raise self.CommandNAKedError
CommandNAKedError
```

ppannuto, 2019-11-12:
I don’t think this is an issue with this library. It sounds like for some reason the saleae software can’t handle the command you’re sending at that time. Depending on your needs, perhaps your script can catch the exception and retry the command. 

If you have some evidence that there’s an actual bug in this library, please feel free to reopen this issue. Thanks!

closed 2019-11-12 by ppannuto

drchopper, 2019-11-12:
The following code is not stable. Could you give me an advice how to get them stable?

I have an issue which occures randomly (sometimes once in 2000 boots). Catching a and retry would lose maybe the 1 failed boot. I want to avoid try, expections.

Thanks

```python
#!/usr/bin/env python
import saleae
import time

dir = os.path.dirname(os.path.abspath(__file__))
s = saleae.Saleae()
s.set_capture_seconds(1)

while True:
    file_path = os.path.join(dir, 'mySaleaeFile')
    if os.path.isfile(file_path):
        os.remove(file_path)
    s.capture_to_file(log_file)
    time.sleep(1)
```

ppannuto, 2019-11-12:
Again I do not think this is a problem that this library can solve — I think this is an issue with the Saleae Logic software. All that this library does is send commands to the Logic software. You (or perhaps Saleae) need to diagnose why the Logic software periodically fails to save the file.