ppannuto/python-saleae #54: erratic behavior after calling get_analyzer

flagline, 2020-02-19:
Disclaimer: I just started using this module, so I may not be using it correctly.  

I am using the saleae module in my application to capture serial communication to and from a device.  Before calling capture_start(), I use the get_analyzers() and get_active_channels() calls to verify that the saleae is setup correctly.  The first capture works fine, but subsequent captures raise an exception when calling get_analzyers().  A little debugging showed that the _recv() routine was returning 'TRUE' in response to the 'GET_ANALYZERS' command.  It appears there is some sort or synchronization problem and the get_analzyers() is getting the response to a previous is_processing_complete(). 

My workaround was to use a try block around the get_analyzers() call and loop until successful.  The first call raises an exception (except for first time) that I ignore and the second call is successful.  This resyncs the socket communication and no further problems (so far).

ppannuto, 2020-02-20:
Hmm, I haven't seen anything like that happen before. It certainly doesn't sound like anything that is supposed to happen.

Are you only calling `capture_start` or are you doing something to check when the capture is complete? Could you reduce to some minimum working example that exhibits the bug?

flagline, 2020-02-20:
Here's a test program and output that I wrote which demonstrates what I'm seeing.
`from saleae import Saleae       # version 0.9.0
from time import sleep

#   Saleae setup is:
#       - Logic version 1.2.29
#       - 2 Digital channels only: rx & tx of serial device
#       - Asynch Serial analyzer on each channel: 0 = rx, 1 = tx
#       - Trigger on rising edge of digital channel 0
#       - Capture duration = 1 second
socket = Saleae()

# call is_processing complete() every 100ms until consecutively true for 'min_times'
#   Print T or F after each call to is_processing_complete
def wait(min_times):
    i = 0
    while i < min_times:
        if not socket.is_processing_complete():
            print('F', end='')
            i = 0
        else:
            print('T', end='')
            i += 1
        sleep(0.1)

# Capture and export results 3 times.
for i in range(3):
    print('\nCapture #' + str(i+1))
    while 1:
        try:
            print('Analyzers: ' + str(socket.get_analyzers()))
            break
        except Exception as e:
            print(e)
    print('Channels: ' + str(socket.get_active_channels()))

    print('Capture: ', end='')
    socket.capture_start()
    wait(8)
    print()

    print('Export 1: ', end='')
    socket.export_analyzer(1, 'capture/rx.csv')
    wait(8)
    print()

    print('Export 2: ', end='')
    socket.export_analyzer(2, 'capture/tx.csv')
    wait(8)
    print()
`
[output.txt](https://github.com/ppannuto/python-saleae/files/4231834/output.txt)

flagline, 2020-02-20:
Here's a test program and output that illustrate what I'm seeing.
[SaleaeTest .py.txt](https://github.com/ppannuto/python-saleae/files/4231847/SaleaeTest.py.txt)
[output.txt](https://github.com/ppannuto/python-saleae/files/4231849/output.txt)

ppannuto, 2020-02-24:
Interesting.. it's quite surprising to see any `F` in the Export list. I believe that `wait` should be using `is_analyzer_complete`. Though that still does not explain how things get out of sync. Perhaps the issue stems from not waiting for the first export analyzer to finish before triggering the next?

flagline, 2020-02-25:
I tried setting a breakpoint in _recv (line 242 of saleae.py) so that I could step slowly and allow plenty of time between commands.  It also allows me to see the _rxbuf contents after each command.  But even stepping through the commands slowly, it's behaving oddly.   For example, the calls to is_processing_complete() following start_capture() return 'NAK' the first time, 'ACK' the second time, and 'TRUE/nACK' after that.  The reply on the second call looks incorrect to me.

ppannuto, 2020-02-25:
Hmm.., from that description it sounds like it may be an upstream issue?
Those inconsistent replies from the Logic software would certainly confuse
processing on this end..

flagline, 2020-02-26:
It could be.  I appreciate your input, but I've put in as much time as I can afford looking into this. Despite the glitches, this module has been super helpful.

ppannuto, 2020-02-26:
I understand! Thanks for all your debug effort.

If you find the loop method to be robust enough as a workaround and think it's worth integrating generally in the library, I'm happy to take a PR.

In the meantime, I think I'm going to close this issue out as an upstream bug (attn @Marcus10110). Feel free to re-open (you or others) if new information merits.

closed 2020-02-26 by ppannuto

flagline, 2020-02-26:
No problem.  For what I'm doing, it seems to be working, but I'm only using
a small subset of the commands.   I think for a real solution, you'd need
to identify where the problem is coming from.  I'm still a bit mystified as
to exactly what is going on.