ppannuto/python-saleae #57: enum34 should not be required above Python 3.4

iwane-pl, 2020-04-14:
As Python 3.4 introduced enum module to the standard library, the requirement to install enum34 should be conditional to Python 2.7.

Current situation (unconditional requirement) breaks other modules depending on Enum (e.g. installing PyInstaller on Python 3.6 after installation of Saleae fails with ``'enum' module has no 'IntFlag'`` due to enum34 being imported before standard enum)

Environment: Python 3.6.8, Windows 10

ppannuto, 2020-04-20:
Interesting -- that makes sense. I'm not great at the internals of python packaging, so if you can put together a PR that implements this, that would be amazing, otherwise I'll try to get to it in the next few weeks.

iwane-pl, 2020-04-21:
I'd love to, but I don't have a Saleae Logic available at home and cannot do it at work (IP rights claim from the employer) - the PR would be untested then :(
Here's some information on this: https://hynek.me/articles/conditional-python-dependencies/ 

I'll still have the PR on mind if I manage to get my hands on Saleae setup at home.

PR #60 (2020-07-01): Fix enum34 not required for Python 3.4 and above
Modify requirements in setup.py to not install enum34 when not needed. Reason: enum34 is a backport of Python 3.4 Enum for Python 3.3 and older so installing it for Python 3.4 and above can cause issues with other packages using the enum. Closes #57
commit d5a30e03 (2020-07-01): Fix enum34 not required for Python 3.4 and above

Modify requirements in setup.py to not install enum34 when not needed. Reason: enum34 is a backport of Python 3.4 Enum for Python 3.3 and older so installing it for Python 3.4 and above can cause issues with other packages using the enum. Closes #57
closed 2020-07-01 by ppannuto