ppannuto/python-saleae #83: Logic 2 Automation Interface - Feedback Request

Marcus10110, 2022-07-07:
Hi @ppannuto, and everyone else currently using the Logic 1.x automation API!

This is Mark from Saleae. We're working on a new automation interface for Logic 2, and would love to get your feedback. You can find more information, including samples here:
https://discuss.saleae.com/t/saleae-logic-2-automation-api/1685

We would really appreciate it if you took a look and let us know what you think! I know the Logic 1.x socket API is quite painful to work with directly (thanks again for creating this awesome python library). We're trying to address as much of that as we can in the new interface.

The new interface uses gRPC, and we'll be publishing the proto file for that, as well as a python library of our own, to maximize flexibility while also providing an easy to use python starting place, inspired in part by this library.

ppannuto, 2022-08-15:
Hi @Marcus10110!

I'm excited to see first-class support for scripting coming to Logic2. I took a quick look over the plans and frankly they look very reasonable to me. The biggest pain points were the communication channel, which gRPC should address, and the expectation of interactive operation, which script authors just kind of had to work around. Both seem fixed here.

Truthfully, I haven't used this library in years. I wrote it in grad school to support some long-term testing of chips we were designing. Maintenance has been light enough that I just kind of fold in what makes sense. As a result, I don't think I have too terribly much more insight to contribute to use cases :/.

If it makes sense at all, y'all are welcome to use any of this library in any way (it was always MIT.. just in case, I updated to dual-license MIT/Apache, as that makes lawyers less nervous sometimes).

I'm also happy to relinquish the `saleae` namespace on PyPI to you if you'd like. Though in that case I'd ask that this library be preserved / integrated in whatever you build so that existing scripts don't break (I know my old research group is still using some of the data collection stuff I wrote...). Could just add a `saleae.Saleae2` or `saleae.Logic2` class or something I imagine.

Marcus10110, 2022-08-16:
Thanks @ppannuto! I completely agree about trying to prevent anything from breaking. I'll have the team take a look at the options, to see if there is anything we could do there, but I think we might end up just publishing a different PyPI package name - but we'll be sure to nest the new API a level down, for maximum compatibility.

Thanks for the offer! We're still working on our roll out plan. I'll talk with the team and we'll let you know if we decide to take you up on your offer.
Ee're hoping to have the production version out soon, but we haven't figured out our PyPI plan yet

ppannuto, 2022-08-17:
Sounds good. I really appreciate the Saleae ethos and support for educational hobbyists, happy to help in whatever small way I can.

If it's easier for some of the logistical stuff to move to email, happy to do that too -- I'm pat.pannuto@gmail.com

Thanks again for everything, y'all really built a game-changing product :)